cheapjack/MileCastles #4: Source the RFID cards

cheapjack, 2016-12-19:
Type of card and card reader
Buy some cards & readers to test
Get some advice from @amcewen

amcewen, 2016-12-19:
Readers, I'd generally choose between the [SL030](http://skpang.co.uk/catalog/hf-rfid-module-sl030-33v-i2c-p-1065.html) (and get the [connector cable too](http://skpang.co.uk/catalog/sl030-interface-cable-for-raspberry-pi-p-1222.html) if you're connecting it to a Pi) or the [PN532](https://www.aliexpress.com/item/Free-Shipping-PN532-NFC-RFID-module-V3-NFC-with-Android-phone-extension-of-RFID-provide-Schematic/32271932273.html?spm=2114.01010208.3.60.tz8QtW&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_10084_10083_10080_10082_10081_10060_10061_10062_10056_10055_10054_10059_10099_10078_10079_426_10073_10102_10096_10052_10050_425_10051-10051,searchweb201603_2&btsid=e9846656-d245-422b-aaee-121634e06f7c) (haven't sourced many of those yet, but have been looking at them of late)

Tag-wise, the MiFare 1k can be got cheaply (probably knock-off rather than genuine NXP chips) from ebay, or if you look at NTAG213/216 then [ZipNFC](http://zipnfc.com/) is good.  Let me know if you find any better sources.

cheapjack, 2017-03-06:
I based our original ball park figure of 3000 at 40p a unit to give 3000 which did not include printing so we budgeted for 

Based on Oomph who do custom printing:

|Number|£ Cost inc. VAT|lead time| Cost per Unit
|----|----|----|----
|1000|648|15 days|0.648
|1500|972|15 days|	0.648

cheapjack, 2017-03-06:
Hey just a headsup for @amcewen Ive just realised a weird arduino bluetooth shield that arrived in the DoES post 2 weeks ago was actually my so called [PN532](https://www.aliexpress.com/item/Free-Shipping-PN532-NFC-RFID-module-V3-NFC-with-Android-phone-extension-of-RFID-provide-Schematic/32271932273.html?spm=2114.01010208.3.60.tz8QtW&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_10084_10083_10080_10082_10081_10060_10061_10062_10056_10055_10054_10059_10099_10078_10079_426_10073_10102_10096_10052_10050_425_10051-10051,searchweb201603_2&btsid=e9846656-d245-422b-aaee-121634e06f7c) order from AliExpress so this doesn't inspire confidence!

cheapjack, 2017-04-05:
@cefn Pretty sure we are using [these now](http://www.electrodragon.com/product/mifare-rc522-rfid-card-readerdetector-ic-card/) so I think we are best going with the cheaper RFID tags with MileCastles stickers of our little roman character @gkboulter made. So lets just go with that and get 1000 to start with

cheapjack, 2017-04-06:
Closing as we've chosen cheaper tags (with stickers) that will work [with this reader](http://www.electrodragon.com/product/mifare-rc522-rfid-card-readerdetector-ic-card/)

closed 2017-04-06 by cheapjack