cheapjack/MileCastles #11: Make prototype stories for venues that want to get involved with content

cheapjack, 2017-01-22:
After surveying sites we need to write simple stories based on initial conversations, some people more involved than others.

Site | People | Subject | Done | Staff Making Content?
----- | ----- | ----- | ----- | -----
Arbeia | Alex | Commanders Stables/Victor the groom | <ul><li>[ ] </li> | <ul><li>[ ] </li>
Segedunum | Trevor | Life in the cavalry / Horse sfx | <ul><li>[ ] </li> | <ul><li>[ ] </li>
Great North Museum | Joanne | Needs Follow up email | <ul><li>[ ] </li> | <ul><li>[ ] </li>
Corbridge | Frances & Kevin | Horse paraphenalia / Artist Apprenticeship | <ul><li>[x] </li>
Housesteads | Frances & Kevin | Not sure / Vast 5 acre site | <ul><li>[ ] </li> | <ul><li>[x] </li>
Chesters Roman Fort | Frances & Kevin | *Not using this site* | <ul><li>[x] </li> | <ul><li>[ ] </li>
Roman Vindolanda | Barbara Birley | Agriculture / Retirement / Shepherds | <ul><li>[x] </li> | <ul><li>[ ] </li>
Roman Army Museum | Barbara Birley | Military? | <ul><li>[ ] </li> | <ul><li>[x] </li> 
Tullie House | Tim & team | Trojan Games / Complex |  <ul><li>[x] </li>
Senhouse Maryport | Jane Laskey | Epona / Libations, Religion & the Coastal Frontier | <ul><li>[ ] </li> | <ul><li>[x] </li>

cheapjack, 2017-01-22:
Prototype the stories in Twine

gkboulter, 2017-01-23:
Created a quick summary of site visits and notes on story themes/ideas:
[google drive link](https://docs.google.com/spreadsheets/d/1O9EbYnrRbLq14Vwc4btC5YH3rWIDOSFfX5Bne3iuhpo/edit?usp=sharing)

cheapjack, 2017-01-27:
We've got some material in gdrive now
[A Simple Story Spreadsheet](https://docs.google.com/spreadsheets/d/1wpaTXbh_3mRi3xCvBE2t_YKRgHbxEAwHyVIlOl4Fk4k/edit?usp=sharing) to get people going
A more comprehensive [Prototype Story Sheet](http://bit.ly/MileCastlesStory) and a [Site Information sheet](http://bit.ly/MileCastleSiteInfo)

gkboulter, 2017-02-27:
Have made a very basic Twine game for Corbridge based on Frances' story.
[Twine file and html file are here:]
(https://drive.google.com/drive/folders/0B18v6qTGYbxYRFM0WFNhNWlJWEk)
Has two different endings at the minute based on if you sharpen your tools.
I got the currency variable working but need to build in some fail states when you drop below zero / run out of money. 
It's laid out to be read left to right and passages are numbered per box (eg 1.2 = box 1, text 2).
Can work on it some more once I know if it looks possible in JSON or other format.

closed 2017-03-06 by cheapjack