cheapjack/MileCastles #19: Design principles - No choices at the same box

cefn, 2017-03-02:
Navigating through the example Twine, (which is excellent thanks, @gkboulter ) it's made me think also of some things which could be improved by following a handful of design principles, which I'll raise as single issues. 

What does anyone think of the following...

* Choices should never include the same box
    * Pros
        + This avoids the possible failure of mistakenly tapping twice and skipping to the next game
        + It allows us the option to always provides 'view pages again' at the same box, especially when there are multiple pages which a group might need to re-read/consider
        + It avoids the potential distortion of story expansion from people always clicking on the box which is right in front of them for convenience
    - Cons
       + It means stories have to be designed with a close coupling between 'legs' between boxes, and the relevant content next to a box
        + It limits the number of choices at each junction point to 3 (assuming we have 4 boxes per venue)

gkboulter, 2017-03-02:
Some notes and feedback from phone conversation with @cefn earlier today:

More on design principles:
- Refer to boxes as places rather than numbers. 
Eg at Corbridge, Box 2 is the Stonemason's Workshop / Box 4 is the Merchant's Shop
- Set an ideal number of passages (e.g. Corbridge is around 55 at the minute)
- Set an average number of characters per passage in relation to how much can fit on to chosen screen size at once

Things to check for:
- Passages that link to non-existent passages
- Passages that link to passages outside of the current individual story / museum (if this becomes an issue). 
- Flag up incorrect ID formatting for passages?
- Flag up incorrect formatting when using variables / conditions?

Passage types:
- Page
- Choice (3 options, current box should not be an option)
- Conditional - routes to more than one passage based on a value or condition
- Wait (not implemented in prototype). Used to give people time to carry out physical task, examine cabinets etc.). 

Things we might use the Sack for:
- Currency / denarii
- God points / Roman karma
- TIme of day / time limited actions
- Specific objects collected
- To note game outcomes - e.g. Fail / Pass / Success (provide a score across all games?)

Possible useful elements for visual rendering:
- It would be useful to see the passages represented either as a Grid or as a List (see screenshot from Inform 'World Index').
- Some way of showing the passages in correct order of play
- A representation of links between passages 
- Colours / labels to clearly differentiate between which box each passage belongs to.
- Labels to identify type of passage (page / choice / conditional etc)
- List of 'sack' variables

[image]

cheapjack, 2017-03-02:
Sorry missed call and feedback on all this today. In middle of something then got pulled into a tonne of family stuff and other work. All really good stuff. Will digest and respond tomorrow properly. 

Despite cons of *"Choices should never include the same box"* wonder if this is a good idea. Promotes movement around space despite losing some choice or being able to choose to wait...

More tomorrow thanks both!!

cheapjack, 2017-03-06:
Sorry my very late comments on this

| More on design principles:|Comment
|----|----
| Refer to boxes as places rather than numbers. Eg at Corbridge, Box 2 is the Stonemason's Workshop / Box 4 is the Merchant's Shop | Yes but still like the idea of a roman numeral on the box
| Set an ideal number of passages (e.g. Corbridge is around 55 at the minute) | good idea maybe aim for 60 initially
| Set an average number of characters per passage in relation to how much can fit on to chosen screen size at once | Yes what do you think?

| Things to check/Validate for:
|----
| Passages that link to non-existent passages
| Passages that link to passages outside of the current individual story / museum (if this becomes an issue). 
| Flag up incorrect ID formatting for passages?
| Flag up incorrect formatting when using variables / conditions?
**Added:**
|----
| print the value of a variable `true` `0-255`
| Verify a `passage visited` variable? Alot of prototype used that

|Passage types:| Comment
|-----|----
| Page
| Choice (3 options, current box should not be an option) | I can see that it could cause confusion so agree with @cefn on the "No choices at the same box" and that someone could casually choose the current box whereas if you always have to go somewhere else it avoids this & promote exploration. 
**Added:**
| Conditional - routes to more than one passage based on a value or condition glenn used <<visited("PassageNumber")>>
| Wait (not implemented in prototype). Used to give people time to carry out physical task, examine cabinets etc.).| You dont really need to wait; you can always say "wait by the blacksmiths" or "go wait in the workshop" 

|Things we might use the Sack for:
|----
| Currency / denarii
| God points / Roman karma
| TIme of day / time limited actions
| Specific objects collected
| To note favourable game outcomes - e.g. Fail / Pass / Success (provide a score across all games?)
**Added:**
| a $visitedpassage variable as above wonder if this is a sack item then rather than a type of passage

closed 2017-03-17 by cheapjack