cheapjack/MileCastles #21: Make a video of card activating a screen

cheapjack, 2017-03-06:
@cefn do you think you'd be able to make a short video/gif of a card activating the black on white screen you showed us in the Tullie House cafe? Would you mind doing that for Wednesday.Then we can send to Lisa / management team

cefn, 2017-03-06:
I will do what I can, but had until now prioritised the story modelling with the aim of ensuring you have a framework for authoring. 

For this reason, although Wednesday is a reasonable target and I will begin working on it, I cannot be sure I won't hit short-lived technical issues wiring both together through the cockle. 

I feel it would be a bad idea to invest engineering time in a parallel technology (e.g. using Arduino) for no reason, unless it's super-quick to do, and it wouldn't mean much anyway.

cheapjack, 2017-03-06:
Agree, focus on the story framework but see if theres a quick screenshot of the 40x4 screen possible great but only if that makes sense. Dont want to divert!

We are going to send a cleaned up version of the Corbridge story to them as a deliverable. Dont want to make something for showing thats not useful down the line to the project. If we can get an authoring framework for Thursday or Friday that will really help get that going I think thats still a priority, then hardware end of the week/next 
thanks!!

closed 2017-03-17 by cheapjack