cheapjack/MileCastles #24: "Arrival" text inconsistent with interaction model - suggests feature - fallback behaviour

cefn, 2017-03-06:
The reference Twine story includes the following text. "Welcome to Milecastles at Corbridge!\nGet your registration card from the till and tap the symbol below to begin!" 

Although this makes sense in Twine which has a context for knowing where you are in the story before you click anything, I can't quite make sense of how it should be presented through our interaction model. 

The start box is just the first box you encounter, and it doesn't know you are coming to the game new as you walk up to it, or indeed that anyone is approaching at all (you might be approaching the box as stage 7 in a partially completed game, and it has had no interaction with you yet as you haven't tapped). 

I think the implication or suggestion might be that the box should show something when it is not being interacted with, but this remains true through all boxes at all times, when there has been no recent interaction, and should be designed separately.

Would be good to get a specification from @cheapjack  and @gkboulter as to what this fallback behaviour should be.

cheapjack, 2017-03-06:
Maybe this should really be in the print theres not a real reason for it to be on the 'first' box I think just hangover from early thinking. If they are seeing the message they already have a card!

cefn, 2017-03-06:
Although the screens are free advertising.

cheapjack, 2017-03-06:
True! Just dont want to add anything we dont really need... can you chat on phone for a moment?

cheapjack, 2017-03-06:
I'll work on this agree we need something after conversation earlier

cheapjack, 2017-03-28:
Think this is getting covered in #30 and #37 now so closing for now

closed 2017-03-28 by cheapjack