cheapjack/MileCastles #25: Get Quote for Additional Print to include in Project report

cheapjack, 2017-03-06:


cheapjack, 2017-03-06:
I think @gkboulter said about £900 for 1000 folded double sided print on a nice paper stock board will need this to be accurate to approve I think

gkboulter, 2017-03-06:
**Option 2:**
_Printed Oyster card wallets w/ instruction card insert_

Oyster Card holders / full colour on outer / 2 inner pockets
1000 @ £0.42 = £420 / £504 inc. VAT
1500 @ £0.42 = £630 / £756 inc. VAT 
Quote from Fluid Branding:
[Quote site link](http://www.fluidbranding.com/oyster-card-ticket-wallet-1287808377.html?currency=GBP&gclid=COPk27q1wtICFQ-NGwodphAM_w)

_Business card inserts w/ instructions_
Standard size / colour both sides / recycled board / rounded corners
1000 cards @  = £64.49 /  inc. VAT £77.39
2000 cards @ £0.05 = £ 108.54 / inc. VAT £130.25
[Quote from printed.com
](https://www.printed.com/products/8/business-cards)

**Total cost:**
1000 wallets and cards - £504 + £77.39 + £20 delivery = **£601.39**
1500 wallets and 2000 cards - £756 + £130.25 + £20 delivery = **£906.25**

gkboulter, 2017-03-06:
**Option 1**
_Printed A5 sheet folded into quarters to hold card_

_A5 Sheets / colour both sides / on 250gsm / recycled paper_
1000 sheets - £262.45 / inc. VAT £314.94
1500 sheets - £379.95  / inc. VAT £455.94
[Quote from Printed.com
](https://www.printed.com/products/3/folded-leaflets)

_Glue Dots to mount cards (5000 peelable)_
£31 + £8 delivery  = £39
[Quote from Tapes Direct
](https://www.tapes-direct.co.uk/index.php?main_page=product_info&cPath=136&products_id=806&zenid=3t1b9vhsvq0dtkh04h4b3kc2b6)

**Total cost:**
1000 sheets / 5000 glue dots - £314.94 + £31 + £20 delivery = **£365.94**
1500 sheets / 5000 glue dots - £455.94 + £31 + £20 delivery = **£506.94**

**Needs assembling!** 
The printer quoted will only do one fold so would need a day to add a second fold to all the sheets...
Can we add one or two days assembly at £100 / £150 per day?
E.g
1000 sheets / 5000 glue dots - £314.94 + £31 + £20 + £300 assembly = **£665.94**
1500 sheets / 5000 glue dots - £455.94 + £31 + £20 + £300 assembly = **£806.94**

gkboulter, 2017-03-06:
Self folding (aiming at £1 per card) to sell for £2

Cards are 65p / print should 35p

gkboulter, 2017-03-06:
**Compasses:**

Alternatively these (successfully copying in Glenn this time) - they could be glued to the RFID cards, making for quite a different kind of experience, solving the need for mapping/stickers perhaps (instructions like walk north west to station Q) as well as suggesting navigation between the Milecastles.

[quote 
](https://www.aliexpress.com/item/100pcs-Pack-Clear-Liquid-filled-Button-Small-Mini-Survival-Compass-for-Outdoor-Sports-Camping-Hiking-FLQ177/32594601295.html?spm=2114.01010208.3.113.52MNFr&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10068_10000074_10000032_119_10000025_10000029_10000028_10060_10000067_10062_10056_10055_10000062_10054_10059_10099_10000022_10000013_10103_10102_10000016_10096_10000019_10000056_10000059_10052_10053_10107_10050_10106_10051_10000053_10000007_10000050_10084_10083_10000047_10080_10082_10081_10110_10111_10112_10113_10114_10115_10000041_10000044_10078_10079_10000038_10073_10000035_10122_10121,searchweb201603_10,afswitch_2,single_sort_1_price_asc&btsid=0c367945-0796-4a65-8749-3cbf95a8f10e)

gkboulter, 2017-03-06:
**Option 4**
_A6 Postcards - could add glue dots for staff to assemble(?)_

_A6 / colour both sides / 350gsm matt_
1000 x **0.11** = £ 107.10 / inc VAT £128.52 + £12 delivery / **Total £140.52**
1500 x **0.10** = £154.22 / inc VAT £185.06 + £12 delivery / **Total £197.06**
[Quote from printed.com](https://www.printed.com/products/5/postcards)

Optional: Peelable glue dots @ £30 for 5,000 (see above for link)

cheapjack, 2017-03-06:
Overview of extra print and RFID cost (providing the NXP Mifare 1K is right)

| Name |Number|£ Cost inc. VAT|lead time| Cost per Unit
|----|----|----|----|----
|RFID Card Printed|1000|648|15 days|0.648
|RFID Card Printed|1500|972|15 days|	0.648
|Size 210 x 210 2xside print|1000|289.26|inc delivery|0.29
|Size 210 x 210 2xside print|1000|433.89|inc delivery|0.29

cheapjack, 2017-03-07:
Overview of extra print and RFID cost (providing the NXP Mifare 1K is right)

| Name |Number|£ Cost inc. VAT|lead time| Cost per Unit
|----|----|----|----|----
|RFID Card Printed|1000|648|15 days|0.648
|RFID Card Printed|1500|972|15 days|	0.648
|Size 210 x 210 2xside print|1000|289.26|inc delivery|0.29
|Size 210 x 210 2xside print|1000|433.89|inc delivery|0.29

So basically subsequent card and print would be £0.94 a unit but ive not included compasses in this

closed 2017-03-17 by cheapjack