cheapjack/MileCastles #34: Waterproof Containers

cefn, 2017-03-27:
The one I bought has since escalated in shipping cost and should not be purchased unless the seller agrees a variation on shipping cost and reprices it before you pay...
https://www.aliexpress.com/item/80-250-85-mm-Size-Newest-And-Hotest-PC-Clear-Cover-Waterproof-Switch-Box-Waterproof-Enclosures/32516362724.html?spm=2114.13010608.0.0.Bwmu8g

However it is a DS-AT-0825 and available from other suppliers https://www.aliexpress.com/w/wholesale-DS%25252dAT%25252d0825.html?spm=2114.01010208.0.0.cXn2ac&initiative_id=SB_20170327095542&site=glo&SortType=price_asc&g=y&SearchText=DS-AT-0825

There are other models with different sizes like...

https://www.aliexpress.com/item/DS-AT-0811-80-110-70mm-2014-Newest-Large-IP66-ABS-Waterproof-Switch-Box-IP66-Screw/2055624079.html?spm=2114.01010208.3.197.cmBP76&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10055_10054_10059_10531_10099_10530_10103_10102_10096_10052_10144_10053_10050_10107_10142_10051_10143_10526_10529_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_130_10037_10078_10079_10077_10073_10070_10122_10123_10124,searchweb201603_3,afswitch_1,ppcSwitch_5,single_sort_0_default&btsid=eaf2de66-ce26-4833-80d1-26428ac9791d&algo_expid=2e97d08f-b508-4f3d-9064-669055501e3e-24&algo_pvid=2e97d08f-b508-4f3d-9064-669055501e3e

...or...

https://www.aliexpress.com/item/158mmx90mmx60mm-Wire-Connect-Waterproof-Sealed-Switch-Junction-Box-Clear-Gray/32480361405.html?spm=2114.01010208.3.140.TJVOBv&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10055_10054_10059_10531_10099_10530_10103_10102_10096_10052_10144_10053_10050_10107_10142_10051_10143_10526_10529_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_130_10037_10078_10079_10077_10073_10070_10122_10123_10124-10530,searchweb201603_3,afswitch_1,ppcSwitch_5,single_sort_0_default&btsid=8ce2d38b-a91c-4f94-aa71-30acd4b7f8c5&algo_expid=4485747e-96bb-4e43-a218-52f9170e60cc-16&algo_pvid=4485747e-96bb-4e43-a218-52f9170e60cc

You might find other useful model numbers to search for by scrolling through this multi-page search...
https://www.aliexpress.com/wholesale?site=glo&g=y&SortType=price_asc&SearchText=saipwell+box&page=3&initiative_id=SB_20170327095844&needQuery=n

I was curious to know if a switch cover like the ones linked below could work for activation of a battery-operated device by lifting, and would fall again by default and close the switch when you released it, assuming the build was in a suitable box, and required a tap on the front - people could open it and then release it - a Normally Closed reed switch and magnet might be feasible, perhaps to eliminate friction in the switch...
https://www.aliexpress.com/item/Transparent-Blue-Waterproof-Box-For-Switch-Socket/32716079907.html?spm=2114.01010208.3.270.u81b8l&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10055_10054_10059_10531_10099_10530_10103_10102_10096_10052_10144_10053_10050_10107_10142_10051_10143_10526_10529_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_130_10037_10078_10079_10077_10073_10070_10122_10123_10124-10530,searchweb201603_3,afswitch_1,ppcSwitch_5,single_sort_0_price_asc&btsid=46a453bb-e37c-471e-b6af-65a652ae9609&algo_expid=5208d710-d1d8-4685-95bb-4046e913315c-32&algo_pvid=5208d710-d1d8-4685-95bb-4046e913315c

https://www.aliexpress.com/item/Home-Mounted-Switching-Power-Supply-Panel-86-Mounted-White-Waterproof-Socket-Flip-Waterproof-Box-Panel-PC/32789801690.html?spm=2114.01010208.3.105.8F9pfJ&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10130_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10055_10054_10059_10531_10099_10530_10103_10102_10096_10052_10144_10053_10050_10107_10142_10051_10143_10526_10529_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_130_10037_10078_10079_10077_10073_10070_10122_10123_10124-10531,searchweb201603_3,afswitch_1,ppcSwitch_5,single_sort_0_price_asc&btsid=9a964ce2-fd09-4289-819c-68d58d7fed23&algo_expid=69c8b7d2-2070-4f29-9919-c15865e761fc-13&algo_pvid=69c8b7d2-2070-4f29-9919-c15865e761fc

cheapjack, 2017-03-28:
Sorry didnt checkout github this morning.. durr
I found
This [IP66 Box of good dimensions](https://www.aliexpress.com/item/Low-shipping-cost-high-quality-ip66-plastic-enclosure-box-with-transparent-lid-140-170-95mm/1708264367.html?spm=2114.13010208.99999999.269.VkekmJ) 170mmx140mmx95mm (W,D,H) at $17.44

cheapjack, 2017-03-28:
I've been looking at this size W 140mm x Depth 170mm x Height 95mm: it fits everything in nicely with flexibility etc. 

Also [this explaination of IP ratings](http://www.dsmt.com/resources/ip-rating-chart/) was useful

cheapjack, 2017-03-29:
Heres the footprint of components to scale using [this box's outer dimensions](https://www.aliexpress.com/item/Low-shipping-cost-high-quality-ip66-plastic-enclosure-box-with-transparent-lid-140-170-95mm/1708264367.html) **W 140mm x Depth 170mm x Height 95mm**

[image]

cheapjack, 2017-03-29:
Worth noting could not find datasheet for this box, so Ive made generous inner dimensions going on images on the item page

cheapjack, 2017-03-29:
Obviously the RFID reader would be positioned below the screen ive just moved it to the side so you can see measurements clearly. All measurements pretty accurately measured with calipers (apart from box)

cheapjack, 2017-03-29:
Should be plenty of clearance to drill a grommet for the power input on the bottom side and can see how headers can be arranged sensibly. Will laser cut a net to position components correctly, use threaded/nuts/spacers/bolts to keep it all vertically aligned also. Can see some threaded mounting points internally inside [this box](https://www.aliexpress.com/item/Low-shipping-cost-high-quality-ip66-plastic-enclosure-box-with-transparent-lid-140-170-95mm/1708264367.html), so at least one or 2 threaded points would be a good idea internally for any box we choose.

cheapjack, 2017-04-06:
Lets use [this one](https://www.aliexpress.com/item/Low-shipping-cost-high-quality-ip66-plastic-enclosure-box-with-transparent-lid-140-170-95mm/1708264367.html)

closed 2017-05-18 by cheapjack