cheapjack/MileCastles #35: Get Outdoor Battery power components and solutions

cheapjack, 2017-03-28:


cheapjack, 2017-03-28:
This [panel mount usb](https://www.aliexpress.com/item/New-Brand-Micro-USB-B-panel-mount-male-to-female-extension-cable-M-F-USB2-0/32663445804.html?spm=2114.10010108.1000014.1.4zhjCy&scm=1007.13338.78123.000000000000000&pvid=8c478ab0-b235-4600-9d60-d27842836579&tpp=1) is ok for connecting up power into the IP66 Box or this much [cheaper one](https://www.aliexpress.com/item/1PCS-lot-150CM-OBD-II-OBD2-16-Pin-Male-to-Female-Diagnostic-Extension-Cable/1482186322.html?spm=2114.13010208.99999999.262.Cq2Cm7) which has a space for silicon sealing

cheapjack, 2017-03-28:
I bought and tested [this external non outdoors battery](http://www.ebay.co.uk/itm/232210843823) and seems similar to [these](http://www.ebay.co.uk/itm/Portable-External-12000mAh-USB-Power-Bank-Battery-Charger-For-Mobile-Phone-Tab/351942901523)

cheapjack, 2017-03-28:
This is similar to [the waterproof solar battery](http://www.ebay.co.uk/itm/100000mAh-Solar-Power-Bank-USB-Portable-Battery-Charger-Panel-for-iPhone7-iPad/162285808334) I tested earlier (it powers on connection, turns off when disconnected)

cheapjack, 2017-03-28:
@cefn [this looks even better](http://www.ebay.co.uk/itm/50000mAh-USB-Portable-External-Battery-Solar-Power-Bank-Dual-Charger-High-Power/122383844777) but untested

cheapjack, 2017-03-29:
[This?](https://www.aliexpress.com/item/BYLYND-Solar-Power-Bank-20000mAh-Double-USB-Solar-charger-External-Battery-Portable-Charger-Bateria-Externa-Pack/32751232253.html?spm=2114.01010208.3.1.PfdCSW&ws_ab_test=searchweb0_0,searchweb201602_2_10152_10065_10068_10130_10150_10136_10137_10060_10138_10062_10141_203_10056_10055_10054_10059_10099_10103_10102_10096_10144_10052_10053_10142_10107_10050_10143_10051_10525_10529_10084_10083_10080_10082_10081_10110_10111_10112_10113_130_10114_10516_10515_10037_10078_10079_10077_10073_10070_10122_10123_10124-10150_10152_10516,searchweb201603_1,afswitch_1,ppcSwitch_5,single_sort_0_default&btsid=44650578-7974-43c0-be80-25146f12667f&algo_expid=dade4e72-bf6c-411e-8da1-b79e596618f9-0&algo_pvid=dade4e72-bf6c-411e-8da1-b79e596618f9)

cefn, 2017-03-29:
Reference [https://www.aliexpress.com/item/BYLYND-Solar-Power-Bank-20000mAh-Double-USB-Solar-charger-External-Battery-Portable-Charger-Bateria-Externa-Pack/32751232253.html](the suggested screw-fit female USB micro socket) Is there any reason not to just use a regular cable from the inside of the box to the outside (e.g there's a Male A-type which comes out of the box). I am guessing we will have to improvise extensions in many cases anyway depending on where the power supply is, so I'm not sure what advantage this gives us, relative to simply attaching a retainer to the cable so it can't be pulled out?

closed 2017-04-06 by cheapjack