cheapjack/MileCastles #39: Write Final Games in avatap

gkboulter, 2017-03-30:
Final 6 museums / games to write:
**Corbridge
Housesteads
Arbeia
Segedunum
Senhouse
Tullie House**

Aiming to complete first drafts of:
**Corbridge / Housesteads / Segedunum** week (ending 31/3/17)
Then **Tullie House / Arbeia** week ending 7/4/17
Finalise / checking **all** during week ending 14/4/17
also can work with Ross on **Senhouse** if help needed

cheapjack, 2017-03-30:
Great, also Barbara confirmed they do not want to pursue vindolanda on Monday will respond to her tomorrow and give an update to Lisa. Been focussing on boxes.

cheapjack, 2017-04-12:
@gkboulter email to me I mean :)

gkboulter, 2017-04-12:
sent them over yesterday to you - will send again now :)

cheapjack, 2017-04-12:
Just seen you've already sent them sorry email hadn't refreshed 👍

gkboulter, 2017-04-19:
**Tullie House - Box placements** 

@cheapjack have used your original box definitions (**I - fort, II - arena entrance, III - arena**) plus a fourth box which at the minute is the **IV - garrison** (base area).

Map using the positions we penciled in during visit - all next to power points:

[image]

**I - **FORT**** South wall at gallery entrance immediately to right of door (rideable horse is in this area too)
**II - **ARENA ENTRANCE**** South wall / middle section 
**III - ARENA**: South wall / Projection area
**IV - GARRISON:** North wall / middle section

Will send over game in progress this am, just fixing some bits.

gkboulter, 2017-05-02:
Making a final check through of **Corbridge** and **Housesteads** games today ready for Thursday install. @cefn - could you advise whether I need to make any changes other than fixing bugs etc? (eg reducing no. of passages)

cheapjack, 2017-05-03:
@gkboulter just send me the game files as they are now so I can push them to my repo @cefn can build an integration test image tonight for tomorrows assembly

closed 2017-05-18 by cheapjack