cheapjack/MileCastles #41: Design and order printed maps

gkboulter, 2017-04-04:
Glenn to design layout week of 10th April. 
Order no later than Weds 19th to receive in time for install.
Current price from Printed.com
Size 210 x 210 2xside colour print 	1000  £289.26 	inc delivery 	0.29 p/unit

gkboulter, 2017-04-04:
Content to include:

- Cover image / sprite(s)
- Simple diagram of participating museums along wall
- What is Milecastles?
- How to play inc. how to tap correctly diagram
- Image of box design so players know what to look for
- Link to website (if applicable)
- Credits
- Logos / Hadrian's branding (from Lisa)

gkboulter, 2017-04-05:
Will design stickers for RFID cards in same time frame as above with maps.

cheapjack, 2017-04-06:
How to play as family
How to use outdoor boxes lift up lid
Leave out schools

cheapjack, 2017-04-07:
[image]
Sticker size ref, 23mm diameter sticker would fit within inner groove with about 0.25mm to spare

gkboulter, 2017-04-12:
Sticker previews - thinking the cropped face detail might have most impact as it's quite a small area to see the full figure or face. Plus full figure will be on the map...

23mm diameter sticker options:

[image]
[image]
[image]
[image]

cheapjack, 2017-04-12:
Yeah i like the cropped face best esp. if the full figure in print will look at that in a mo trying to debug a weird thing in arebia I found; I love the game though!

[image]

gkboulter, 2017-04-12:
Link to Map text document - feel free to make edits / changes
https://docs.google.com/document/d/17_eXBXt8rOXjsexcWzBKKhLQ35fpZFGv4Q1RKYhkbaA/edit?usp=sharing
I need to get final design signed off by Lisa as soon as we're happy with layout - pdf to follow this pm.

gkboulter, 2017-04-12:
Map designs so far -   still need to:
- Redraw the map in 8 bit style (current one is placeholder)
- Get rid of some outlines etc on images and logos. 
- Add other facial expressions from animated soldier
- Finalise text
- Finalise box image with @cheapjack if any major changes
- Spin top half of front page 180degrees so that it reads like a booklet when folded

[image]
[image]

cheapjack, 2017-04-12:
Lovely design Glenn! Have updated text in the doc so changing cards to tags

- Show some kind of drawing of tag (with arrow?) being placed against a box in **WHAT IS MILECASTLES**
- In **WHERE TO PLAY** Some kind of helpful drawing? Tap diagram in extreme close-up?

Reference for [box image](https://github.com/cheapjack/MileCastles/blob/master/boxes/BoxForMap.pdf)

gkboulter, 2017-04-12:
Stickers ordered - got 2,000 units in the end as min order was £50.
Waterproof/scratchproof/high tack
Arriving Tues/Weds - will bring to Liverpool on Weds/Thurs.

gkboulter, 2017-04-13:
Updated design - sent to Lisa for approval. Aiming to send to print today as soon as it comes back. 

Added:
- Tapping Instructions
- New wall map w/ key in same panel
- New box image
- Updated text from google doc (trimmed for space in a couple of places)

Still to do:
- Clean edges of Unesco logo
- Final proof reading of text

[image]
[image]

gkboulter, 2017-04-13:
Ordered 100 x maps and 20 square display panel images so that we have some for next week.
Arriving with GB on Weds so will bring over to @cheapjack on Thurs.
Will order full 1,000 units when Lisa has signed off. 

Latest proofs:

[image]
[image]
[image]

gkboulter, 2017-04-19:
- Printed maps approved by Lisa Keys 19/4/17
- GB to show to proofs to @cheapjack and @cefn for feedback at meeting on 20/4/17
- Look at ways of mounting the display panel/image
- Suggest we go with a slightly lighter weight paper for the maps to make folding easier.
- Aiming to order full print run on 21/4/17
- GB to apply stickers and fold batches of maps in prep for each installation

gkboulter, 2017-04-21:
Updated text document for MAP with changes from yesterday's meeting:
https://docs.google.com/document/d/17_eXBXt8rOXjsexcWzBKKhLQ35fpZFGv4Q1RKYhkbaA/edit?usp=sharing

To do before final print run:
- @cefn and @cheapjack to comment on instruction text
- GB to change text in final layout
- GB to add 'horse' tap area to box images

cheapjack, 2017-04-22:
Im not convined by a horse logo to represent the interaction I think some kind of wifi/nfc related thing is more universal. Ive pushed a milestone back for this to 1st May to get final print together. Also @hwayounggit had some suggestions so I'm adding her to the project so she can help with this, documentation of builds and installs etc. She's coming with us to Corbridge on the 5th

gkboulter, 2017-04-24:
Just checked and we need to complete any print changes by **Thurs 28/ Fri 29** to receive the order in time to take to Corbridge on the 5th May. The bank holiday is adding an extra day onto order times again(!). I'll also need to do some folding on 4th so that we aren't trying to do it on the day.

hwayounggit, 2017-04-24:
Print Comments:
**Overall** 
- Need red dotted lines? Seems unnecessary also causes nightmare when folding and not aligned.
- Can the the typeface for the logo - the Roman looking font be a thicker weight? At least for the title. Seems a bit slight
**How to Play page**
Wonder if the How to Play should be on the right and What is Milecastles should be on the left...
- point ii - to read Box I (Roman Numberals) instead of #1 - can be confusing.
- point iii - 'Tap' is in quotes and is 'read point'. Think Tap should be **bold** or some other way of distinguishing the text from 'read' point'. Two different things
- iv - e.g. 'Go to Box IV' should be in TextTextFont.
**What is**
- Can the sample yellow boxes be a bright yellow, like the one on the 'How to Tap' yellow?
**Where to play**
- the circles with the i, ii, iii etc can they be little forts AND change the font from TeleText to something else. It's confusing. Think adding forts will reinforce the idea that it's Hadrians Wall. Didn't get that. Love the little bitmap centurion 
- the red panel with the information looks funny being aligned right. Any reason for that? Would align to the full stops after the numerals.
**How to Tap**
- Don't think you need the : after How To Tap
- Can the circle with the centurion sticker be the shape of the fob?
**Tips and hints**
- Feel like you need a direct instruction/tip in the yellow and then the text that expands. For example: i Play as a group(yellow) We recommend etc...
ii Persevere (yellow) If you don't ...
iii Keep your tag (yellow) Go to any of the six ...
iv Be gentle (yellow) Pressing harder on the boxes will not help you the goddess Minerva will curse you! 
**Credits**
- Check the Hadrian's Calvary logo. Bit snipped on the right.

cefn, 2017-04-24:
I've been posting changes into the doc and extensively commented. Hopefully that radically shortens text for the punchy intro, and also has a suggestion for re-allocating the 'what is milecastles' panel to be the graphical intro (e.g. the moment you open, on the left you see punchy text instructions, on the right you see the corresponding imagery).

gkboulter, 2017-04-25:
Thanks for the feedback @hwayounggit - all really useful!!
Have updated all the text and added changes above - just need to add the updated 'tag circle' box image from @cheapjack then will share.
Aiming to go to print end of day tomorrow / 25th April.

cheapjack, 2017-04-25:
Glenn I've added the [BoxForMap.pdf](https://github.com/cheapjack/MileCastles/blob/master/boxes/BoxForMap.pdf) Ref, look out though ive not got a vector version of the roman face so ive got a placed png object in the pdf. Updated to have the original logo not infilled but still present you may want to remove so it wont glitch the print process when you export for print.

I'll go through and resolve the comments in the [gdoc](https://docs.google.com/document/d/17_eXBXt8rOXjsexcWzBKKhLQ35fpZFGv4Q1RKYhkbaA/edit?usp=sharing) tomorrow

cheapjack, 2017-04-25:
Should be able to proof and resolve everything tomorrow.

gkboulter, 2017-04-25:
Thanks @cheapjack will add the new box images in tonight and preview the new layout here.

gkboulter, 2017-04-25:
Latest proof, with:
- Text updated from gdoc, 
- Changes suggested by Hwa Young 
- New box images with tag circle

Front:

[image]

Front (flipped)

[image]

Back:

[image]

cheapjack, 2017-04-27:
additional comments for @gkboulter to sign off his print
- Remove the space between `Domestic` and `Science.org.uk` so its more like a webaddress (which it is of course)
- Keep 'USING YOUR TAG' over 'HOW TO TAP'
- Change role-play to ROLE-PLAY capitals to refer to the capitals in the game?
- Hwa Young suggested back colour a deeper royal blue to match to the blue in the boxes? to be honest Im not sure the print will look as nice and a direct match could spoil it. We just couldnt find enough acrylic to match the blue. Worth noting the blue in the boxes will be almost totally hidden in most cases so I think the current colour scheme which is not a match is fine. Image below shows difference and i dont think matching them will give much benefit.

[image]

cheapjack, 2017-04-27:
Sorry for being so late @gkboulter with this! Cant easily see where a diagram of lifting the lid can be incorporated. Currently considering a tinted acrylic lid so you can still see the same 'skin'.
Ive suggested adding `LIFT LID on boxes OUTDOORS` and removing the parantheses underneath point `iii` in the HOW TO PLAY panel which I think covers this. Capitals in yellow capital font as in the rest

cheapjack, 2017-04-27:
Ill be in Morecambe at 1030 so should we do quick final proof then? I can talk on the journey up over bluetooth.

gkboulter, 2017-04-27:
Just making changes now - any time from 10:30 good to do quick proof over phone!

cefn, 2017-04-27:
I had hoped that given How to Play includes the relevant _text_ guidance, that the panel immediately to the right (currently titled 'Using your Tag') was redundant in that form, meaning the panel could be dedicated to graphical depictions of the parts of the system. That's 60% of what it is doing already, but by dropping the "Using your Tag" element and the guidance text, and instead focusing on graphical depictions you could potentially have a pictorial 'frame' showing placing the tag, a second 'frame' depicting text on screen, and a frame depicting outdoor lid opening, possibly a fourth frame for anything else we want to depict (is there a way to depict mime? that's a challenge for you!). 

I'm really shy of doing a big print run at this stage. We haven't even nailed the final interaction sequence, which will only be finally settled by a few days of testing, (e.g. do you remove to wait for the text or not), yet we're baking our provisional implementation into potentially a large number of leaflets. 

I fear we may regret it, given a disjoint between the instructions and the actuality could be messy and need explanation, defeating the purpose of the leaflet. Requiring ourselves to design the interaction to match the leaflet is very heavily the wrong way round.

gkboulter, 2017-04-27:
Some stickered tags for reference:

[image]

cefn, 2017-04-27:
They look just right! Based on my rule of thumb of 2x Foundation Living Wage for manual labour activities and the estimate of time taken for 100 you gave today, (unless you find a much more efficient way of sticking them), we should charge the sticking at about £16.90 per 100, on top of the sourcing and printing. Roughly 17 pence each. Could some kind of laser-cut jig help, which can accept e.g. 50 tags in the right orientation, making the alignment+sticking part easier?

gkboulter, 2017-05-02:
Latest batch of print received (400 copies of map) for Crobridge / Housesteads install on Fri 5th. 
Will fold as many as possible before Fri - lower paper weight folds much more easily.

gkboulter, 2017-05-05:
GB to add final instructions in response to play test (eg loading times / safety advice on tag as choke hazard) on **Monday 8th May.** to receive final print run in time for installs.

gkboulter, 2017-06-02:
Latest draft of print design.

Have updated:
- About / Prize draw text (prize tbc)
- Map key with instructions to check off museums
- Hints and Tips - added choke hazard warning

[image]

- Instruction text
- How to use tag - text update (image being updated now

[image]

cheapjack, 2017-06-02:
## HOW TO PLAY
change TAG CIRCLE to TAG SYMBOL
LIFT TAG or REPLACE TAG to LIFT TAG or PLACE TAG message

## USING YOUR TAG
PLACE your tag 
for a few seconds to
load game from tag 
until LIFT or PLACE 
message appears
?

gkboulter, 2017-06-02:
Updated maps (aiming to print early next week if possible) - would be good to get your feedback @cefn!

[image]

[image]

cefn, 2017-06-02:
I've separated the instructions about gameplay and tags in this draft. Hope it's helpful and not two steps back. The text below is based on the latest interaction and capitalises only the imperative phrases. It's designed to make it feel 

# HOW TO PLAY

i) FIND BOX 1 and PLACE YOUR TAG flat on tag symbol
ii) READ PAGE, MIMING ANY ACTIONS (no cheating!)
iii) FOLLOW INSTRUCTIONS on-screen
iv) DISCOVER YOUR ENDING
v) PLAY AGAIN to discover different endings

# USING YOUR TAG

i) LIFT LID for outdoor boxes
ii) PLACE TAG on tag symbol to load a page
iii) HOLD IN PLACE until bookmark saved
iv) REMOVE TAG when finished reading the page
iv) REPEAT ii) iii) iv) until choice page presented
v) GO TO BOX matching your chosen action

gkboulter, 2017-06-05:
Thanks @cefn  - am updating the layout today

cheapjack, 2017-06-05:
Just for consistency here's the proof of those updates @gkboulter did tonight

[image]
[image]
[image]

cefn, 2017-06-05:
These look totally brilliant! Well done @gkboulter !

Only one last tweaky comment. I think the TIPS AND HINTS section could easily drop the i) ii) iii) metaphor since they are not naturally ordered and it would allow punchier presentation (more whitespace, more text or bigger text in that box), as well as enabling the i) ii) iii) metaphor to have better impact elsewhere. At present I think it struggles a little with the multiple indentation and duplicate color heirarchy as per the original How to Play but nowhere near as important to fix, given these tips are a 'bonus' panel.

cheapjack, 2017-06-06:
Yes I'd agree on that.

gkboulter, 2017-06-07:
Lisa has suggested she can offer a DS as a prize - should we still push for Microbits?

[image]

Updated Hints and Tips section:

[image]

cefn, 2017-06-07:
Anything will be fine. However, a DS sounds better doesn't it (and more in theme with young gaming enthusiasts). Perhaps Microbits as runner up. Can we park this with weasel text in the guide, like "amazing prizes" which might make people go to the URL anyway to find out?

gkboulter, 2017-06-12:
Updated prize text - 

[image]

cheapjack, 2017-06-12:
Updated prize text looks good to me!

gkboulter, 2017-06-12:
Final run - ordered 1200 copies (200 per site) - £209
Arriving Weds 14th June with Glenn

gkboulter, 2017-07-06:
Updated maps to correct Housesteads / Corbridge mix up.
200 x copies sent to Corbridge for Monday 10th July
GB to order 800 new maps for remaining 4 installs and fold in time for 17/18th Juy.

closed 2017-07-09 by cheapjack