cheapjack/MileCastles #46: Confirm numbers of RFID tags to match sticker print run

cheapjack, 2017-04-09:


gkboulter, 2017-04-12:
Planning to order 1500 x 23mm or 20mm diameter stickers to go with **1,000** RFID tags.
It's cheaper to order 500 spare stickers now (£5 or so) in case we need further tags later on.

cheapjack, 2017-04-12:
Cool! Id go with the 1500 x 23mm which is a perfect fit for the tag I got from @cefn that i referenced [here](https://github.com/cheapjack/MileCastles/issues/41#issuecomment-292514955)

cheapjack, 2017-04-12:
double checked 23mm diameter is a good fit for that tag 👍

gkboulter, 2017-04-12:
Going with a different printer for these that does custom sizes as existing one (printed.com) will only do 25mm. These will do custom size next day delivery: http://www.discountstickerprinting.co.uk with anti scratch lamination for approx £25-£30

cheapjack, 2017-04-12:
Lamination just might give it that nice gift shop keyring feel so Id go with that. If we were printing RFID cards the print costs on that would have been much bigger so this is still cheap

gkboulter, 2017-04-24:
We have **1900** stickers on hand so will cover the initial 1,000 tags plus a top up of 900 if we need to restock the tags at a later date.

closed 2017-06-27 by cheapjack