cheapjack/MileCastles #57: Confirm the text fonts for cefn

cheapjack, 2017-04-17:
[image]

I think [tim08](https://github.com/cheapjack/MileCastles/blob/master/screens/fonts/timR08.png) should be the main text. I've mocked up its spacings above in a 5x7 font I had which is a good fit. Worth noting that capitalisation takes up more than 5x7. 8x7 in most cases except for capital `M` which is 9x7. As a result, @gkboulter when using capitals for the ACTING out text I would allow for this with extra character spacings. In the above case its fine

I think use [timB12](https://github.com/cheapjack/MileCastles/blob/master/screens/fonts/timB12.png) for the extra big first landing text described in #37 and the idle text described in #30

For TAP TO CONTINUE & Goto III and other capitals inversed as part of the interaction text described [here](https://github.com/cheapjack/MileCastles/issues/29#issuecomment-287317741) we'd still use the capitals in [tim08](https://github.com/cheapjack/MileCastles/blob/master/screens/fonts/timR08.png)

cheapjack, 2017-04-17:
Picked tim08 because it looks nice and makes me think of times new roman and also is right size to give us those 9 lines of 26 characters we've been working to, although we'll need to allow for the capitals taking up more horizontal pixels

cefn, 2017-04-18:
Not sure that mocking up with a different font will give you much of an idea how it presents I'm afraid. To test this out you'd have to use the font itself. You could use and tweak the Pillow render routine of https://github.com/ShrimpingIt/bitfont to validate pixel-for-pixel.

cefn, 2017-04-18:
This is what it looks like...

[image]

cefn, 2017-04-18:
However, it comes out at approx 9 pixels high, which is why we need to check against the real fonts. You can't have TimR08 and still hit the dimensions you're hoping for.

cefn, 2017-04-18:
By comparison, this is the font_5x7 which I've been using for testing - 7 pixels high even with the proper line spacing...

[image]

cefn, 2017-04-18:
How many pixels high are you expecting individual characters to be in the 'extra big' text? Based on the examples I saw, it would be 2 lines high, therefore 32 pixels. For that reason it's not the best idea to use a 12 pixel high font I feel, even drawn with double pixels.

There are others which are bigger and will render at a large size. Alternatively something which is a divisor of 32 (e.g. 16 or 8) would be able to be scaled to fill and be large as possible, though I don't have any scaling routines yet.

```helvB08``` for example is actually 11 pixels high, meaning it could scale to 33 pixels high. ```ncenB10``` renders at 18 high, but is actuall 16 plus a lot of space above. So far the ideal one, though, seems to be ```ncen24B```. I made a test render for CAPS only, (which cuts off the descenders from the lower case letters). It is exactly 32 pixels high! See below...

[image]

cefn, 2017-04-18:
Having said all that, I have a fear that our memory limits will force us to avoid high-res fonts, so scaling could be the only way to go. Hitting a brick wall on memory at the mo. Everything else is go.

cheapjack, 2017-04-19:
Ok scrub my suggestions no worries;  if memory is solved and we can use these fonts can we just use the `font_5x7` you mention you've used for testing [above](https://github.com/cheapjack/MileCastles/issues/57#issuecomment-294735834) and then the `ncen24B` for caps?

closed 2017-05-05 by cheapjack