cheapjack/MileCastles #63: Draw a horse for the boxes

cheapjack, 2017-04-20:


cefn, 2017-04-21:
[image]

cheapjack, 2017-04-22:
Actually im not convinced a horse is right now. And its not to do with Trump! It makes sense in the game world but I *think* generally kids of the welded to the tablet generation are pretty aware of a wifi/wireless logo...

cefn, 2017-04-22:
If I recall, we weren't trying to find an image to represent the interaction, and in fact that was the key takeaway from this part of the meeting. 

The issue was rather what can we INSTRUCT people to do. Horse is good for this, but the main conclusion we had drawn was 'anything but the wifi/tag logo' since it was near-impossible to instruct without a paragraph of description, as hinted by your use of 'wifi/wireless logo'. What is that vector art called? We don't want to be saying RFID and it means little anyway. Perhaps you have a solution for this which means we _can_ use the RFID/WIFI logo but somehow solve the original issue of instructing? Horse has absolutely no problem here.

We are not expecting anyone to know what to do without following the instructions and the semantics are entirely unlike Wifi. The only possible advantage of some already-laden symbol is if you were to hand someone the tag and say 'go play the game' with no instructions, seems an unlikely scenario.

gkboulter, 2017-04-24:
Tried and struggled to make a horse image on Friday that was in keeping with the aesthetic...

Could we go with something really simple and universal like a circle / square / compass points?

cheapjack, 2017-04-24:
[image]

Yes that was the logic we ended up with anything but a rfid contactless sign i guess..I'd lost the rationale for it: ive also tried to make a lo-bit horse so its in keeping but not happy withit:
compass is a great idea though and as 'tapping' is how you 'travel' through the game it kind of makes sense... @gkboulter can you knock up a compass in keeping? Remember must be cut from single piece of acrylic..

cheapjack, 2017-04-24:
Felix suggested putting the stickers on the tap point. I think would rather it a physical part of the box rather than a sticker which feels a bit cheap... you do have to press the tag flat to the horse to get a read so will have to add that to the text instruction..

closed 2017-05-03 by cheapjack