cheapjack/MileCastles #68: Make a website frontpage pointing to milecastles.uk

cheapjack, 2017-04-22:
Include
Video in-use
All material in the additional print

gkboulter, 2017-06-30:
Soldier graphics for website are in this folder:
https://drive.google.com/open?id=0B5imPjsSmpRkTEw5dU81Nk1id3c
Google form for prize draw is here:
https://goo.gl/forms/bqOiZGwvtmmz0e493
Jupiter font in there also
Will replace map jpgs once we finalise next (final) run.

cheapjack, 2017-06-30:
Initial barebones [homepage for milecastles.uk url](http://milecastles.uk) done. Will improve, add instructional artwork and proper fonts, logos, tips, blog posts etc over wknd 

Will leave this open until we happy with it and we've made logos and colours better

thanks @gkboulter !

gkboulter, 2017-06-30:
Looks really good Ross - great that it's up in time Tina Turma weekend. I wouldn't rush to add anything else to it before next week now the front page and survey link are up.

cheapjack, 2017-06-30:
👍 Also updated the Domestic Science homepage just in case `Tiny Turma` commissions us for a 10 million quid text adventure based on `Mad Max Beyond the Tinderdome`

closed 2017-07-09 by cheapjack