cheapjack/MileCastles #71: Resolve memory issues to upload a senhouse game in time to install on Tuesday

cheapjack, 2017-04-23:
Jayne Laskey at Senhouse in Maryport gave us her preferred option of Tuesday 25th April to install, and also Thursday 28th is possible.

I've got 3 boxes wired up ready for flashing, awaiting their numerals vinyl and horse logo (or whatever), but we'll need to make a decision on whether its still possible to meet the Tuesday date by early afternoon Monday so there's time to respond to Jane reasonably and bump to thursday which feels more likely at the moment.

@cefn can you please let us know whether memory issues resolved by early afternoon Monday 23rd? We'd need to load up senhouse.py in my fork for that install of course, i said focus on corbridge because that seeems a fairly big one (4 boxes) in terms of memory

cefn, 2017-04-24:
Sorry not to realise the schedule brought back up (Tuesday! aargh). I've made the memory-saving changes ready for testing. Need to attend a short meeting at 9:30 and then can look into testing and verifying the interactions work with the full stack during the day tomorrow.

I am still aiming for Monday delivery of the working system, but given it's a hard deadline and still plenty to verify, (including identifying any significant changes needed via interactive testing as we discussed) a Tuesday install is definitely best avoided.

cheapjack, 2017-04-24:
Cool let's catch up in the afternoon and I'll bump to Thursday :)

cheapjack, 2017-04-24:
Sorry lets catchup in  the morning, pottering on git, got the integration test success brain bit frazzled for phone :) be good to try putting the full stack on the boxes tomorrow if we've got it now we've verified the wiring :)