cheapjack/MileCastles #72: Senhouse trigger/completion logic

cefn, 2017-04-23:
Just going through Senhouse. 

In the following...

```python
        page = """
        {% if node.change.triggered %}
            {% if node.change.completed %}
            You offer grain from the
            store. What is he
            supposed to do with this?
            He's not Bruno Mars!
            DANCE like Bruno Mars!
            {% else %}
            Mars is bored of grain
            He encourages harvests
            but its no fun!
            {% endif %}
        {% else %}
        Mars is satisfied
        {% endif %}
        """
```

I don't think that "Mars is bored of grain" can ever be triggered. It is only arrived at in the case that the change is _triggered_ but not _completed_. Triggering but not completing arrives only when a _minus_ attribute for a named sack variable is provided, but there isn't enough of that sack variable to take away (for example money). BTW, this behaviour of not completing changes is configured by the change's stayPositive attribute - true by default).

cefn, 2017-04-24:
Another four such offenses should be taken into account :) That's because minus isn't used anywhere in the story. An example might be if you had a limited amount of wine, and you can run out.

Search for "node.change.completed"

cheapjack, 2017-04-24:
Ive took out offending `node.change.completed` elements and additional `{% else %}` thanks for checking!

closed 2017-04-24 by cheapjack