cheapjack/MileCastles #78: Put a "Shrimping inside" logo on the boxes

cheapjack, 2017-04-29:
@cefn you got your shrimping logo anywhere as an `.svg` or some kind of vector format? Couldnt seem to find in the ShrimpingIt assets in git ;)

cheapjack, 2017-04-29:
Its fine if you dont want a logo on by the way, and you dont have to have the Intel inside reference ;)

cefn, 2017-05-03:
The ShrimpingIt logo in vector/stencil style is at https://github.com/ShrimpingIt/projects/blob/master/vector/logo_stencil.svg 

You might want to remove the word 'Learn' for clarity, and just have "Prototyping" perhaps, or something else you want when cutting/stickering. The file is edited in Inkscape and in that environment you should be able to ungroup and then edit by placing a cursor in the curved text in text mode (F8), it remains as text bound to an invisible arc.