cheapjack/MileCastles #85: Basic outdoor box running instructions

cheapjack, 2017-05-03:
- Outdoor box is attached with an external solar battery 
- LIfting the box lid activates the box ready for play, closing the box sends it back to sleep
- External solar battery will need to be periodically swapped every 2 days/weekly with the identical battery stored in the office
- Carefully disconnect the battery after opening its waterproof bag
- Connect with fresh battery and reseal the bag 
- Use only the supplied power supply, to recharge batteries, it has a raspberry logo and a Milecastles sticker on it

cheapjack, 2017-06-23:
@gkboulter we need this for Monday 26th June for Frances to review

gkboulter, 2017-06-23:
@cheapjack will do this today/Monday and send to Frances for Monday noon. Will make a general version that can be adapted for sites with mixture of indoor/outdoor boxes.

gkboulter, 2017-06-26:
Started doc here: 
[Google doc](https://docs.google.com/document/d/1LmOioHl9tQV6nZiNMnaV2-PxuIDCnLINOb3pa8PIB9g/edit?usp=sharing)
@cheapjack - Just need advice on a couple of points before sending to Frances.
Can print some copies also and attach map for reference.

cheapjack, 2017-06-26:
Ive updated it, resolved comments and sent as pdf to you, Lisa & Frances ta 👍

gkboulter, 2017-06-26:
Thanks Ross - am in Barrow tomorrow so can print a few copies ready for Weds.

cheapjack, 2017-06-27:
Made a final update to pdf after revised outdoor box behaviour emailed to Corbridge and cc'd @gkboulter 👍 Need to do a similar one for Tullie but without the outdoor box bits if you can do that?

closed 2017-07-03 by cheapjack