cheapjack/MileCastles #86: Provide integration test image and wiring reference

cefn, 2017-05-03:
This is an issue as per telecon with Ross to provide an integration testing image as well as wiring reference information for the use of Hardware SPI.

cefn, 2017-05-03:
The initial version of the firmware image, tested locally with my wiring is now available at https://github.com/cefn/avatap/blob/master/firmware/integration-test.bin

cefn, 2017-05-03:
Wiring is as per the Hardware SPI pin assignments at https://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html#hardware-spi-bus

I have assigned colors based on 20 pins - half a 40 pin rainbow dupont split exactly down the middle. 

_NOTE!_ The cockle is placed in a breadboard with an offset, to allow two pins next to the D1-D8 pins. This means two wires need to be inserted into the 5v and GND rows to break out the rows to the power rails on the other side beneath the cockle BEFORE inserting it in the breadboard. I used a pair of tinned copper (@ShrimpingIt) jumper wires pre-bent to come out from under the board at the top, bent back and connected to the power rails.  

The looms have the following sequence, in two groups of 6 for the screen and one group of 8 for the reader. They are listed below in the sequence left to right based on the boards facing upwards flat on the desk, with pins pointing toward you. note the 10 colors are each used twice...

## Screen Loom 1

Brown, Red, Black, White, Grey, Purple 

Far left 6 pins

## Screen Loom 2
Blue, Green, Yellow, Orange, [Red, Brown TWISTED] !! Keeps Red +5v, Brown GND

Far right 6 pins

## Reader Loom 

Black, White, Purple, Blue, Green, Yellow, Orange

All 8 pins

# Loom to Cockle wiring details

## Screen Loom 1  

BLK -> 0V [Brown] - Backlight Cathode
BLA -> 5V [Red] - Backlight Anode
VOUT -> UNCONNECTED [Black] - Reference voltage
RST -> 5V [White] - Screen Reset
NC -> UNCONNECTED [Grey] -  Not Connected (unused pin)
PSB -> 0V [Purple] - Parallel Serial Bus selector (indicating we are using Serial - SPI)

## Screen Loom 2

E-> D5/GPIO14 [Blue] - SPI SCK (Serial Clock)
R/W -> D7/GPIO13 [Green] - SPI MOSI
RS -> D8/GPIO15 [Yellow] - Screen SPI Slave Select
V0 -> 5V [Orange] - LCD Contrast voltage
VCC -> 5V [Red] - Power supply +5V positive
GND -> GND [Brown] - Power supply common negative

## Reader Loom 

SDA -> D4/GPIO2 (Black) - Reader SPI Slave Select 
SCK -> D5/GPIO14 (White) - SPI Serial Clock
MOSI -> D7/GPIO13 (Grey) - SPI MOSI
MISO -> D6/GPIO12 (Purple) - SPI MISO
IRQ -> UNCONNECTED (Blue) - Interrupt Request Pin (as yet unused) 
GND -> GND (Green) - Power supply common negative 
RST -> D3/GPIO0 (Yellow) - Reader Reset 
3.3V -> 3.3V (Orange) - Low Power supply +3.3V positive

cheapjack, 2017-05-04:
@cefn Integration test is working so if you can get a Corbridge game image together tonight I think its worth us going up and scoping out the install and get some documentation. Magnet mechanism a little bit dodgy but im working on that.

cefn, 2017-05-04:
Shall I just designate a pin as the Polulu OFF pin? Should we say D1 (at a guess) since D0 seems to be special around powerup.

cheapjack, 2017-05-04:
👍 yes D1 should do

cheapjack, 2017-05-17:
@cefn Ive actually been wiring D2 to OFF on the pololu now sorry

cefn, 2017-06-14:
Now superceded by https://github.com/cheapjack/MileCastles/issues/107

closed 2017-06-14 by cefn