cheapjack/MileCastles #87: Provide Scale-Test Images for Prototype Corbridge Boxes

cefn, 2017-05-04:
There are four largely untested images for the four corbridge boxes at...
https://github.com/cefn/avatap/tree/master/firmware

These are identical to the box1 tester image I showed in the demo video, but with a different hard-coded 'box identity' configured on boot for each case. 

Blank tags are expected to be overwritten with Corbridge-specific story storyUid, nodeUid and sack data if no previous data is found on them. However, once written, they should be recognised as part-played Corbridge tags. Be careful with them, therefore. If we change the logic, we might end up with half-played tags which create inconsistencies or break running boxes in the future. For now, let's try not to mix them with future customer tags.

Currently the repo hasn't been updated with all the source changes needed to build the images, but at least the images are there.

cefn, 2017-05-04:
BTW I have no idea if this is the latest corbridge. It's based on https://github.com/cefn/avatap/commits/125f1c3a52aa9180b3fc9d227e117498ac701c31/python/stories/corbridge.py

cheapjack, 2017-05-06:
Got images on boxes i - iii some glitches and  seemingly slow reads of tags but Need to review wiring thoroughly and list all observations. It would make sense if @cefn came over to DoES on Monday or Tuesday and troubleshoot with the actual boxes wired up and in use and maybe with Julian if he's around I'll email him; I'm working on something else Monday with Adrian but you could work in the workshop... Then I'm working on boxes Tuesday and Wednesday when I'll have to try build and wire up more boxes (a further 5 indoor and 3 outdoor) while we try resolve glitches in time fort next install on Thursday. It's awesome to see the game almost there though! it feels really nice! But we do need some kind of feedback to a successful read I think aside from displaying the text the game wants...

cheapjack, 2017-06-23:
as we've made new images for corbridge & tullie and remaining glitches have been broken out into seperate issues I'm closing this

closed 2017-06-23 by cheapjack