cheapjack/MileCastles #96: Modify the Place Tag Below font

cheapjack, 2017-05-18:
Agree we need this to be bigger, just want to tweak font, message and justify centre..
Be nice if it varied somehow but think thats a non-essential enhancement at the moment

```
PLACE TAG
  TO PLAY
```
or maybe if smaller font (but still big)
```
  PLACE TAG
TO CONTINUE
```
like the direction of below but then the print kinda covers that.

cheapjack, 2017-05-19:
# Working out our instructional text...
### Keep tag placed while reading or you will miss it
`WAIT`

### Make sure we read the tag and write to it
`LOADING 
 GAME TAG`

### Hold tag until writing is finished
`SAVING GAME...`

### Remove When ready
` READY TO
REMOVE`
### Replace for the next page

Taking 
Remove and replace to get the next page

cheapjack, 2017-05-23:
` HOLD TAG
   BELOW`
alternating with
`XXXXXXXXX

MILECASTLES
 
XXXXXXXXXX`

`Loading Game...`
instead of
`Reading Tag...`

`Saving Game...`
instead of
`Writing Tag`

Once card read & written (loaded & saved) render text delay text by 3 secs?
#### "Now that its loaded and saved you can remove your tag if you've finished reading, and then hold tag below again to get the next page in the game"
`Read Remove and Replace for more`
`Ready to Remove Tag`

#### If we dont save the game to the tag (partial write)
` HOLD TAG
   BELOW`
again

cheapjack, 2017-05-30:
## Final instructions and default behaviours
Based on recent discussions with @cefn and @gkboulter this is where we are at

Default behaviour: dont worry about asci art but we just need something like an arcade 'standy screen' where `HOLD TAG BELOW` is a bit like `INSERT COIN`
So Maybe we get an arcade flashing text of:
```
XXXXXXX
   MILE
CASTLES
XXXXXXX

```
```
XXXXXXX

XXXXXXX

```
Maybe 2 'flashes' alternating with Capitalised, centre justified `font_ncenB14` (will need a new line to prevent chopping off the top) This smaller font is so there is some spacing around the text which just looks nicer in terms of typography (acknowledge the trade off to distance of ledgibility)
``` 

HOLD TAG
   BELOW

```
or
```

HOLD TAG
FOR PAGE

```

On presenting card, 
```

                       Loading Game...
```
then

```
Welcome to Milecastles!
Look for numbered boxes
and hold your tag to
progress the story...

                       Saving Game...
```

Once card read & written (loaded & saved) 

```
Welcome to Milecastles!
Look for numbered boxes
and hold your tag to
progress the story...

       Lift & hold for new page...
```

`absenceDelay = 0`: @cefn warned against this as people may get used to the delay rather than train themselves to just keep the tag on the reader

Then revert to Default behaviour:
```

HOLD TAG
FOR PAGE

```
followed by 2 flashes of
```
XXXXXXX
   MILE
CASTLES
XXXXXXX

```
```
XXXXXXX

XXXXXXX

```

What do we think of that? Im around later to chat 2ish?

cefn, 2017-06-01:
# Synonyms we considered...
* Present
* Place
* Put
* Position
* Touch
* Dock
* Slot
* Insert

Hold sounds like it could mean the opposite. Press suggests you might have to apply force, which we want to avoid for mechanical reasons. I think we've nailed it by trashing the word 'hold' and going with Place/Replace. 

I think we settled on
 
    PLACE TAG
    TO READ

...as the general message, alternating with an occasionally flashing

    MILE
    CASTLES

I really like emphasising Milecastles ambiently, assuming that people can contextualise this phrase, which is otherwise mystifying (e.g. point-of-sale and leaflets have 'Milecastles' prominently). 

The read write page sequence looks like...
1) PLACE TAG TO READ (BIG) [tag expected to be presented by player]
2) Loading game...[ tag kept in place]
3) Saving game... [ tag kept in place]
4) Lift & replace for more [ONLY IF PAGES REMAIN, ELSE NO LABEL] [tag kept in place until finished reading]
5) Place tag to read (SMALL) [ONLY IF PAGES REMAIN, ELSE LOOP TO 1)]

cheapjack, 2017-06-01:
## Final instructions and default behaviours
Updated @gkboulter based on last discussions with @cefn today

Default behaviour:

```

 MILE
CASTLES

```
Flash
```

```
```

 MILE
CASTLES

```
Then Capitalised, centre justified `font_timB14` (will need to prevent chopping off the top) 
[image]

```

PLACE TAG
 TO READ

```

On presenting card, 
```

         Loading Game...
```
then start with slightly more instructional intro text.  

```
Welcome to Milecastles!
Look for numbered boxes!
Place your tag on the
tag symbol below & wait,
then lift & remove
to progress the story...

          Saving Game...
```

Once card read & written (loaded & saved) 

```
Welcome to Milecastles!
Look for numbered boxes!
Place your tag on the
tag symbol below & wait,
then lift & remove
to progress the story...

Lift & replace for more
```

```
Welcome to Milecastles!
Look for numbered boxes!
Place your tag on the
tag symbol below & wait,
then lift & remove
to progress the story...

       Place tag to read
```

`absenceDelay = 0`: @cefn warned against this as people may get used to the delay rather than train themselves to just keep the tag on the reader

Then revert to Default behaviour:

```

PLACE TAG
 TO READ

```
followed by 
```
 
 MILE
CASTLES
 

```
```
 

 
```
```
 
 MILE
CASTLES
 

```

I think that's it!

cheapjack, 2017-06-01:
Just re-confirmed behaviour more visually for @gkboulter & re-opened so he can see, will let @cefn have pleasure of re-closing :)

cefn, 2017-06-02:
Actually we've gone one better. Had a chat with @gkboulter and we now have...

* Keep in place, loading
* Keep in place, saving
* Lift & replace for more
* Place tag to read

closed 2017-06-02 by cefn