cheapjack/MileCastles #98: Complete the games with 24 char limit and add ... to ThroughSequences

cheapjack, 2017-05-23:
## @gkboulter

- [ ] Arbeia 
- [ ] Housesteads
- [ ] Segedunum
- [ ] Tullie House

## @cheapjack

- [ ] Corbridge
- [ ] Senhouse 
- [ ]

gkboulter, 2017-05-24:
@cheapjack - just pushed Housesteads with character limit fixed and game paths tested in new emulator. Arbeia to follow tomorrow am.

gkboulter, 2017-05-25:
@cheapjack pushed updated **Arbeia** with character limit fixed and tested all paths/endings in new emulator. Working on Tullie next.

cheapjack, 2017-05-25:
@gkboulter great ive pulled them to mine. Hit a weird loop in Corbridge where cant seem to get to the pub! Trying to resolve it

gkboulter, 2017-05-26:
Nearly there with Tullie House - need another half day or so to test properly and reduce the text as it's too at the minute in terms of reading/tapping.

gkboulter, 2017-05-31:
Sorry, **Tullie** taking a bit longer than expected and slowed down by couple of problems testing on emulators (not always building all templates), back on it tomorrow a.m. and will push then.

cheapjack, 2017-06-01:
Don't worry Glenn @cefn tied up until this afternoon anyway. I'll plod on with boxes

cheapjack, 2017-06-01:
Make sure we add the more explicit Intro text to the first `landing` ThroughPage:
```
Welcome to Milecastles!
Look for numbered boxes!
Place your tag on the
tag symbol below & wait,
then lift & remove
to progress the story...

```

cefn, 2017-06-04:
I think we need to be as careful with this as the other instructional guidance we have offered, so let's put this text in one place and import it from there, where it can be maintained and improved. Named variables can be referred to just like any other value, so in stories/__init__.py we could declare...

introText="Blah blah"

...and in the actual story files...

page=stories.introText

In particular, "lift and remove" is a typo. 

Also the guidance to 'place below...then lift' is misleading compared to the more closely considered wording we have elsewhere, which should prevent people from lifting until both the following are satisfied...
* game has been saved to the tag
* player has finished reading current page

cheapjack, 2017-06-05:
ok @gkboulter @cefn how about:
```
introText="""
Welcome to Milecastles!
Look for numbered boxes!
Place tag on the symbol 
& keep in place to read
then lift & replace to 
progress the story...
""",
```

and in most `landing` uid's we've got a ThroughSequence to start with so how should we get to it? 
Easiest way seems to be that we make `startNodeUid=firstLanding` then `from stories import introText` and make this point to the `landing` first page; then we can remover the Welcome message from there?
```
ThroughPage(
uid="firstLanding",
goalBoxUid=entranceBox.uid,
nextNodeUid="landing",
page=introText,
)
```

cheapjack, 2017-06-05:
I cant get it to work that way as I dont seem to have a `stories.init.py` and didnt know how to mess with that so Ive just placed introText as a variable at the top

cefn, 2017-06-05:
Stupid Github markup hid the other characters. Should have read ```stories/__init__.py``` see https://github.com/cefn/avatap/blob/master/python/stories/__init__.py it's currently empty but can contain any needed python.

It's nice to have a single copy of this text (given it's embedded inside the actual stories, not part of the guidance system) to satisfy the requirements of DRY -
 from the Pragmatic Programmer. See https://en.wikipedia.org/wiki/Don%27t_repeat_yourself

cheapjack, 2017-06-05:
Yes no worries I've added it now, just not fully up on using `__init__.py` so now it has simply
```
introText="""
Welcome to Milecastles!
Look for numbered boxes!
Place tag on the symbol
& keep in place to read
then lift & replace to
progress the story...
"""
```
Ive added that in as suggested, also done so with TullieHouse.py and will do rest of @gkboulter games as i get them. So ive pushed `TullieHouse.py` and `corbridge.py` so we can build game images for those asap 👍

cheapjack, 2017-06-05:
I'll add the `introText` and `firstLanding` to Segedunum once I get from @gkboulter then we can close this and all games are ready for @cefn to build

I've done some further Tullie and arbeia testing, made some amends added some logic so that characters don't say True and False added the `introText` and `firstLanding` to both and fixed a weird bug in senhouse.

btw should we be using the agnostic module in everything for the cross platform issue? Only appears in Senhouse at present just so you know

cefn, 2017-06-05:
I'm expecting to go through using agnostic garbage collection in a paranoid way to minimise memory fragmentation, but don't worry, that's not your job :)

cheapjack, 2017-06-05:
👍

closed 2017-07-09 by cheapjack