cheapjack/MileCastles #102: Update Lisa Keys on Progress

cheapjack, 2017-06-05:


cheapjack, 2017-06-05:
Be good to update her with proof of the revised print text @gkboulter 👍

gkboulter, 2017-06-05:
Latest proofs with revised text from Friday / Monday - can help with update tomorrow @cheapjack 

As pdf (zipped):

[HC Map v11.pdf.zip](https://github.com/cheapjack/MileCastles/files/1052974/HC.Map.v11.pdf.zip)

As jpgs:

[image]
[image]
[image]

closed 2017-06-07 by cheapjack