cheapjack/MileCastles #104: Review the power-off cycle

cheapjack, 2017-06-05:
Currently using `D2` connected to `OFF` on the Pololu in the outdoor boxes so just need to go over the code for that. Should be included universally I guess but with indoor boxes ignoring it.

Bit mixed up over the resistor required on the Pololu`OFF` is it from `OFF` to `GND IN` on Polulu @cefn ?

cefn, 2017-06-26:
For reference, I propose using D8 connected to OFF on the Polulu.

In the new wiring, D2 is used for one of the SPI wires to the screen, so that won't fly. The only free pins currently are D0/GPIO16 D3/GPIO0 and D8/GPIO15. GPIO16 and GPIO0 might both struggle with Polulu OFF role - GPIO0 has to be pulled up on boot to ensure entry into ```FLASH (NormalRunning)``` mode. I guess the circuitry in place for this would therefore send a high pulse to OFF the moment it powers up. GPIO16 has some behaviours about sending pulses as the Cockle moves out of deep sleep too, so best avoided for removal of doubt even though it's unlikely to be a problem. That leaves D8/GPIO15 which I'll try to prove tomorrow by integrating with the final image. 

# References for my reasoning

You can see some of the special behaviours of Cockle pins according to their GPIO numbers [here](http://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations). It reports that the voltage of D8/GPIO15 helps to determine the boot mode, requiring it to be pulled down in both modes we actually use - Upload and NormalRunning, so that's what I'll try. 

And [this thread](http://www.esp8266.com/viewtopic.php?f=32&t=11614) offers some confirmation...
> Since GPIO15 (D8) is used by the boot mode process, most of the boards have this pin pulled down by an external resistor, which is stronger than internal weak pullup.

Finally the [NodeMCUv2 schematic](https://raw.githubusercontent.com/nodemcu/nodemcu-devkit/master/Documents/NODEMCU_DEVKIT_SCH.png) indicates D8/GPIO15 has the correct pull in place already hard-wired. After boot, we can set D8 it as a GPIO pin with an _active low_ voltage, then pulse it _active high_ to trigger power-off at our leisure.

# Your Circuit question

For reference, there are two entirely different circuits here and your comment seems to suggest they are coupled somehow. 

1. One is a circuit wired across a pair of pins used to control POWER-ON behaviour. A button is normally expected to be wired across these pins, and a reed switch is equivalent to a button. In this sub-circuit there needs to be no wiring to the Cockle, and no wiring of a pull resistor.
2. Entirely separately, there is a wire from a digital pin (D8 proposed) to the POWER-OFF pin of the Polulu board which should be set in software to trigger POWER-OFF when needed.

The [documentation for the Polulu switch](https://www.pololu.com/product/2808) describes operation as...

> OFF	A high pulse (> 1 V) on this pin turns off the switch (e.g. allowing the target device to shut off its own power).

In case the pin attached to OFF was ever floating (e.g. during power up before it is told to be an output and set to a specific voltage) I originally had some concerns that it might trigger POWER-OFF just from random variations. Having selected D8, there is a resistor pull-down even when the microcontroller hasn't actively set the voltage, so fingers crossed this case should be handled automatically.

cefn, 2017-06-27:
Update, D8 is apparently set to value(1) == 3.3V on boot for some mysterious reason (a LED lights until you set it as an output pin in software, when it suddenly goes dark). I fear this means we can't use it.

cheapjack, 2017-06-27:
Thanks for all that @cefn hadn't seen it before we talked.

So we've decided to go with what you describe [above](https://github.com/cheapjack/MileCastles/issues/104#issuecomment-311205945)

"a circuit wired across a pair of pins used to control POWER-ON behaviour. A button is normally expected to be wired across these pins, and a reed switch is equivalent to a button. In this sub-circuit there needs to be no wiring to the Cockle, and no wiring of a pull resistor."

So I've gone with a 10k pull-down resistor between `CTRL` on Pololu and `GND` on Pololu and then the reed switch wired between `5V` & `CTRL` wired in it's normally closed config with the magnet contact keeping it open *ie* When lid is lifted the switch reverts to it's normally closed position and connects `CTRL` to 5V so pulses `HIGH` from  the `LOW` that the pull-down resistor is keeping low. At least that's how I understand it and it works!

So @gkboulter outdoor behaviour is hold lid up to power up and use, then close lid when you're done which seems pretty logical even if you need 2 hands. Dont think this clashes with the print description in any way 👍  

Ill leave this open for reference or if we ever want to return and get auto power off, but lets not do any more work on it for these first installs.

closed 2017-07-03 by cheapjack