cheapjack/MileCastles #107: New Integration Test and Wiring with Reclaimed Pins

cefn, 2017-06-14:
To avoid screen glitch I have wired Reader SPI and Screen SPI separately. This loses some pins.

I have also retired any software-controlled Reset on both boards, and retired SlaveSelect on the screen. This reclaims some pins.

The wiring notes in this issue are intended to supercede https://github.com/cheapjack/MileCastles/issues/86

The screen is wired to rainbow cable - face down with the '20' mark on the right, counting down from right to left - as follows....
* Screen Loom 1 (Far right 6 pins 20-15)
  *  Brown, Red, Black, White, Grey, Purple 
* Screen Loom 2 (Far left 6 pins 1-6)
  * Blue, Green, Yellow, Orange, [Red, Brown TWISTED] !! Keeps Red +5v, Brown GND
* Reader Loom (All 8 pins)
  * Black, White, Purple, Blue, Green, Yellow, Orange

# Loom to Cockle wiring details

## Screen Loom 1  

BLK -> 0V [Brown] - Backlight Cathode
BLA -> 5V [Red] - Backlight Anode
VOUT -> UNCONNECTED [Black] - Reference voltage
RST -> 5V [White] - Screen Reset
NC -> UNCONNECTED [Grey] -  Not Connected (unused pin)
PSB -> 0V [Purple] - Parallel Serial Bus selector (indicating we are using Serial - SPI)

## Screen Loom 2

E-> D2/GPIO4 [Blue] - SPI SCK (Serial Clock)
R/W -> D1/GPIO10 [Green] - SPI MOSI
RS -> 5V [Yellow] - Screen SPI Slave Select
V0 -> 5V [Orange] - LCD Contrast voltage
VCC -> 5V [Red] - Power supply +5V positive
GND -> GND [Brown] - Power supply common negative

## Reader Loom 

SDA -> D4/GPIO2 (Black) - Reader SPI Slave Select 
SCK -> D5/GPIO1 (White) - SPI Serial Clock
MOSI -> D7/GPIO13 (Grey) - SPI MOSI
MISO -> D6/GPIO12 (Purple) - SPI MISO
IRQ -> UNCONNECTED (Blue) - Interrupt Request Pin (as yet unused) 
GND -> GND (Green) - Power supply common negative 
RST -> 3.3V (Yellow) - Reader Reset 
3.3V -> 3.3V (Orange) - Low Power supply +3.3V positive

Have to go to a meeting right now, but I just tested this wiring sequence from scratch and looks fine for new integration.bin image.

cheapjack, 2017-06-14:
👍 wiring up now

cheapjack, 2017-06-14:
Made wiring changes in bold for ease in your comment

cheapjack, 2017-06-14:
upload firmware in /avatap/firmware with
`esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200 write_flash --flash_mode dio --flash_size=32m 0 integration-test.bin`

cefn, 2017-06-14:
The draft images are geared up for responsiveness, not avoiding glitchiness. So at the time of writing they DO NOT refresh the screen every time, demonstrating that most of the time the redraw logic does work. 

The mystifying logic which maps from rectangles to selective blocks of pixels in the ST7920 (and seems to be broken) is the send address and data bit at the bottom of the redraw() method in https://github.com/ShrimpingIt/micropython-st7920/blob/master/st7920.py

Changing the images to redraw the whole screen may be needed if selective writing is not possible.

cheapjack, 2017-06-23:
@cefn just re-wired and tested the Corbridge outdoor box again and the reed switch triggers power up as normal wired into `D2`. However immediately powers back off as if the cockle is triggering the D2; is this because it's used in the reader as SCK? Have we actually got the power off cycle in the current images for the outdoor boxes.

I guess we have `D3` and `D0` free now, should I use one of these? Ive tried `D3` but has same behaviour. Its not anything else bypassing the reed switch is all fine. Ive some time Monday 26th and Tuesday 27th to try sort this out before install.

closed 2017-06-23 by cheapjack

cheapjack, 2017-06-27:
Now removed the auto power  off and Pin juggling see issue #104

cheapjack, 2017-07-09:
# Final Loom to Cockle wiring details *(note bold text are changes required from first wiring spec)*

## Screen Loom 1  

BLK -> 0V [Brown] - Backlight Cathode
BLA -> 5V [Red] - Backlight Anode
VOUT -> UNCONNECTED [Black] - Reference voltage
RST -> 5V [White] - Screen Reset
NC -> UNCONNECTED [Grey] -  Not Connected (unused pin)
PSB -> 0V [Purple] - Parallel Serial Bus selector (indicating we are using Serial - SPI)

## Screen Loom 2

**E-> D2/GPIO4 [Blue] - SPI SCK (Serial Clock)**
**R/W -> D1/GPIO10 [Green] - SPI MOSI**
**RS -> 5V [Yellow] - Screen SPI Slave Select**
V0 -> 5V [Orange] - LCD Contrast voltage
VCC -> 5V [Red] - Power supply +5V positive
GND -> GND [Brown] - Power supply common negative

## Reader Loom 

SDA -> D4/GPIO2 (Black) - Reader SPI Slave Select 
SCK -> D5/GPIO1 (White) - SPI Serial Clock
MOSI -> D7/GPIO13 (Grey) - SPI MOSI
MISO -> D6/GPIO12 (Purple) - SPI MISO
IRQ -> UNCONNECTED (Blue) - Interrupt Request Pin (as yet unused) 
GND -> GND (Green) - Power supply common negative 
**RST -> 3.3V (Yellow) - Reader Reset**
3.3V -> 3.3V (Orange) - Low Power supply +3.3V positive

## Pololu power trip for Outdoor boxes:

![img](https://a.pololu-files.com/picture/0J6761.350.jpg?a49cd139840f29608e58b4b0a77d

10k pull-down resistor between `CTRL` on Pololu and `GND` on Pololu and then the reed switch wired between `5V` & `CTRL` wired in it's normally closed config with the magnet contact keeping it open ie When lid is lifted the switch reverts to it's normally closed position and connects `CTRL` to `5V` so pulses `HIGH` from the `LOW` that the pull-down resistor is keeping low. At least that's how I understand it and it works!