cheapjack/MileCastles #112: Corbridge

cefn, 2017-06-27:
Spotted something unusual in corbridge. 

The second smith1 line will override the first, so I'm sure this is not intended but I don't know what is intended.

```
       hideChoices={
            "smith1": "sack.sharp==True",
            "stone1": "sack.stone==True",
            "smith1": "sack.smithAnger==True",
        },

```

cefn, 2017-06-27:
I am re-engineering corbridge for space BTW so just tell me what is intended and I'll put it into my copy.

cefn, 2017-06-27:
For now I have modified it as follows...

```
        hideChoices={
            "smith1": "sack.sharp==True or sack.smithAnger==True",
            "stone1": "sack.stone==True",
        },

```

gkboulter, 2017-06-27:
That looks correct thanks @cefn - the blacksmith option only shows up the tools are not sharp and you haven't visited him earlier to trigger 'smithAnger'.

closed 2017-06-27 by cheapjack