cheapjack/MileCastles #115: Install at final three/four sites

gkboulter, 2017-07-07:
Just putting this in one place for reference, so I can keep track whilst away.

GB to:
- [ ] Send install docs to Frances / Alex / Jane 
- [ ] Send staff guide docs to Frances / Alex / Jane
- [ ] Get confirmation from Frances on Housesteads for 18th
- [ ] Order 800 or  600 new prints for final sites
- [ ] Fold 600/800 maps for final sites 
- [ ] Finish stickering remaining 800 tags
- [ ] Send approved ARBEIA game to @cheapjack
- [ ] Wait for approval on Housesteads and Segy (due Friday 7th) and check 
- [ ] Send approved HSTEADS / SEGY to @cheapjack 
- [ ] Confirm times with Housesteads / Senhouse for Tues 18th
- [ ] Source cordless drill if @cheapjack not able to borrow from Does
- [ ] Glenn to install Senhouse by train on Weds 19th if Tues 18th not possible

@cefn and @cheapjack :
- [ ] @cefn to collect boxes and all brackets / fixings from @cheapjack 
- [ ] @cefn to flash boxes with final games (SENHOUSE / HOUSTEADS / SEGY / ARBEIA)
- [ ] Glenn to confirm meet up times for 17th /18th installs with @cefn 
- [ ] Glenn and @cefn to install remaining sites on 17th / 18th

cheapjack, 2017-07-09:
Take note of recent comments in issue #114 before re-flashing `corbridge IV`
Drill is from Patrick in DoES so when Ive arranged date to handover boxes to Cefn I will get that drill unless @cefn has his own powerful cordless drill

cheapjack, 2017-07-19:
updated [milecastles webpage](http://milecastles.uk) to reflect latest install/deinstall

closed 2017-08-10 by gkboulter